gkoh/furble #10: Focus Button on m5Stick

dimitrij2k, 2022-06-13:
Is it possible to use the other Key as Focus button? So we could have Focus & Shutter... the M5 have a second button 😉

gkoh, 2022-06-16:
That would be great to support, however ...

There are 3 buttons on the M5StickC:
* A - front button, used for 'OK' and 'Shutter'
* B - left side button, used for 'Next' and 'Back'
* C - power button, not sure we can get to this easily

At the moment, after a user enters the 'shutter release' mode, the 'B' button is used to exit that mode (and A is used to take photos).

There are at least 2 problems:
* need to investigate whether there is a 'focus' command sent over the Bluetooth protocol
* how to exit the 'shutter release' mode if 'B' is used for 'focus'
   * maybe long press? (eg. long press to exit, short press to focus?)
   * happy to discuss ideas

Unless you are referring to the M5Core, which has 3 front buttons?

dimitrij2k, 2022-06-16:
Okay, i see the point… but once we enter the Shutter Mode, does anybody really need to go back? I can hold the power button for power off… 😉 i cant believe, someone go back to connect to a second camera… maybe you could make two versions…

damn, trust mem its time for a donation button *hahaha*…

dimitrij2k, 2022-06-19:
I like the Long-Press for Autofocus idea 👍 but „finding“ the Autofocus command over bluetooth is like magic for me and im not a wizard 😆

gkoh, 2022-06-25:
Hmm, the 'focus' command is going to be interesting.
I played around with the Fujifilm phone app in bluetooth only mode (which is what furble does).
It seems to really have only one button, which when pressed sends the following:
* cmd - shutter
   * cmd - shutter down
and when 'unpressed'
* cmd - shutter
   * cmd - shutter release

When I originally implemented, I was aiming to support 'CL' -> 'continuous low', which on the camera is just hold the shutter down and let it take lots of photos in low speed.

It seems that on 'shutter down' it attempts to focus at that point.
I wonder if I we send multiple 'cmd - shutter' or 'cmd - shutter release' we can just get focus?
Need to investigate/experiment further.

gkoh, 2022-06-25:
> Okay, i see the point… but once we enter the Shutter Mode, does anybody really need to go back? I can hold the power button for power off… wink i cant believe, someone go back to connect to a second camera…

I originally had plans (and still do) to implement:
* long exposure (bulb) mode
   * set a timer on shutter open (eg. 60s exposure)
   * where time is configurable
* intervalometer
   * take _n_ shots with _m_ delay between
   * where _n_ and _m_ are configurable

So, when you enter 'shutter mode', you then get 3 options and can enter one, then exit and choose another.

> damn, trust me… its time for a donation button _hahaha_…

Perhaps, I will look into it :slightly_smiling_face:

closed 2022-11-06 by gkoh