gkoh/furble #9: User-Friendly Instruction...

dimitrij2k, 2022-06-13:
Maybe a more "User Friendly" Tutorial will open this Project to more Users! Its so f**king Awesome what Fujifilm Users can get here!

Like this step-by-step guide.... (just as an idea)

1. Install PlattformIO (Infos here: https://docs.platformio.org/en/latest//core/installation/methods/installer-script.html#super-quick-macos-linux)
2. Download this Code and extract it to the folder.....
3. Plug in your... and turn it on...
4. Start this code in Terminal...

Just a little more Infos ;-)

gkoh, 2022-06-16:
This is a good idea.
I will start a wiki page when I get some extra time.

dimitrij2k, 2022-06-19:
Maybe i can help you, to make a noob friendly (like me) tutorial ;-)

gkoh, 2022-11-11:
Following some discussion in #19, I tested [esphome-flasher](https://github.com/esphome/esphome-flasher) with success.
I was able to download the latest 1.4.0 release to a Windows 10 system and program binary release v1.0.0 to my M5StickC.
One caveat with *esphome-flasher* is it _always_ erases the device, thus wiping all settings including pairing information.

For first time users, and with the very few settings furble has, this is somewhat acceptable.
We will need to explore other options for users wishing to upgrade and preserve settings.

I will update the wiki with instructions for *esphome-flasher*.

gkoh, 2022-11-20:
> I will update the wiki with instructions for _esphome-flasher_.

Done, it is now here:
https://github.com/gkoh/furble/wiki/Easy-Install

closed 2022-11-20 by gkoh