gkoh/furble #14: Rename Project for better finding...

dimitrij2k, 2022-06-25:
Maybe it would be better to find this nice Project, when you rename it from 

**"furble - FUjifilm Remote Bluetooth Low Energy"**

to 

**"furble - Fujifilm Remote Bluetooth Trigger/Button" (Trigger or Button or both)**

I just found the way to furble in an Forum as a link from anybody. With Google search  i cant find a link to furble ;-) 

> Donation button + better finiding in google search = Fame & Money 🤣

gkoh, 2022-06-26:
Thanks for the suggestion!
The project is indexed by Google, but it does lack a few keywords.
(I tried to find it via Google too ... it was very difficult).

 I have updated the 'About' bit to hopefully get on the radar a bit more. I could also add some of the more recent released/popular cameras (eg. X-T3, X-T4, X-S10).
I will keep an eye on this for a little before closing this issue.

I'm curious though, which forum did you find this project in?

gkoh, 2022-08-04:
The updated 'About' now appears to have been indexed by Google.
At least, I can find it if I search with some various terms.
I can also see a slight increase in referring traffic from Google.

closed 2022-08-04 by gkoh