gkoh/furble #28: Shutter releases immediately (Canon)

wolcano, 2022-11-14:
When A or B are pressed (and not released) the camera does auto-focus and releases shutter immediately, not waiting for button (A or B) release.
Tested on Canon EOS-RP

gkoh, 2022-11-15:
Focus only isn't working at all, so let's ignore that here (see #29).

I think we should define how this is expected to work for Canon cameras.
The current implementation is modelled after my Fujifilm setup, which is kinda like this:
* Mode set to continuous low speed
* When shutter is fully depressed
   * Camera keeps taking shots
* When shutter is released
   * Camera stops
* Focus is achieved with classic partial press of shutter

So, with *furble*, this is:
* Button A press -> shutter held down
   * Time passes here, where camera is in continuous shoot mode
   * This will also support bulb mode, where the shutter is held open for as long as the button is down
* Button A release -> shutter release
* Button B press -> shutter partial
* Button B release -> shutter release
   * With my Fujifilm, if I don't send this command, the camera is stuck in focus mode

gkoh, 2022-11-15:
Thus, @wolcano, what behaviour would you like to experience for the Canon's?
We are free to change how the remote responds to:
* Button A press
   * Something happens whilst pressed?
* Button A release

gkoh, 2022-11-20:
> When A or B are pressed (and not released) the camera does auto-focus and releases shutter immediately, not waiting for button (A or B) release. Tested on Canon EOS-RP

I just read this properly, please disregard my question @wolcano. I need to get access to the M6 again to work out what's going on.

gkoh, 2022-11-22:
Managed to get the M6 for a little bit.
I can confirm the Canon app is still terrible and works exactly the same as reported here, pressing the shutter button immediately triggers and there is no focus button.

So, `furble` is consistent with the app :slightly_frowning_face: 

Looking at https://github.com/robot9706/CanonBLEIntervalometer, they've found the pairing exchange has a 'mode' indicator.
As our reverse engineering was based off an Android phone, `furble` is sending `0x02` for type.

I think we need to send 0x03 to emulate the BR-E1 and get access to the other functions.
Not sure if the M6 mark 1 I have will support this (online search says it may not), we'll see.

closed 2025-01-31 by gkoh