gkoh/furble #29: Implement focus support for Canon EOS

gkoh, 2022-11-15:
Canon EOS support for focus during shutter control doesn't behave as expected.
Investigate how or whether it is possible to bring up to parity with the Fujifilm cameras.

gkoh, 2022-11-22:
Based on [this discussion](https://github.com/gkoh/furble/discussions/31), we will need to implement the 'Remote' service handling to get focus support.
I do not have access to any newer EOS cameras, but I can guess what may work as a starting point.

hijae, 2025-01-27:
I can test R6 Mark2 for a long time, and RP for a short time(1weaks?). Is there a commit or branch to test?

hijae, 2025-01-27:
OK. I'll check it out and let you know the results.

hijae, 2025-01-27:
I tested it with both devices and was able to pair in remote control mode, not a smartphone mode.

gkoh, 2025-01-27:
> I tested it with both devices and was able to pair in remote control mode, not a smartphone mode.

Excellent, thank you very much for the effort.
I should have enough information from the various sources to implement remote support, meaning:
* proper focus
* proper shutter

Eventually we may need to support smartphone mode to supply GPS geotag, but that can wait until basic remote mode works properly.

Leave this with me for a week or two.

gkoh, 2025-01-31:
Much thanks to @hijae for testing this on Canon EOS R6 Mark II and Canon EOS RP.