gkoh/furble #41: furble identifies as a smartphone with new Fujifilm firmware

gkoh, 2022-12-15:
Newer cameras that are support the new tripod remote (TG-BT1) have new firmware (eg. 1.50 for X-T30).
With this firmware, when `furble` connects, a little smart phone icon appears under the Bluetooth icon.
This is expected as `furble` is built from replicating the protocol between the smart phone app and the camera.

It would nice if we could work out the TG-BT1 protocol as that is more in-line with `furble`.

josephschmitt, 2023-01-11:
Just wanted to say that I bought an M5StickC Plus on a whim after finding this project and just got it all working perfectly with my X-T5 with literally a single command. This is such great work, and I can't believe Fuji hasn't created a BT wireless shutter release accessory yet. Thanks for getting this going!

gkoh, 2023-01-18:
> 

@josephschmitt Thanks for the note! That's another (different and quite new) confirmed working model.

gkoh, 2023-08-20:
Closing this issue as "won't fix".
I'd like to clean up and complete the GPS work in #43, which requires `furble` to mimic a smartphone.

closed 2023-08-20 by gkoh