gkoh/furble #46: Remote will not work with new Fujifilm updates

adrianuseless, 2023-07-14:
I have heard from other users who have updated their Fujifilm cameras that the remote has stopped working

Can this be corrected so to will work?

Adrian

gkoh, 2023-07-18:
Thanks for the report.
As noted in the [discussion](https://github.com/gkoh/furble/discussions/45), this is a known issue.
However, I need to find time to analyse the changes in the latest firmware.

One of the challenges is once I update my X-T30, I can no longer test the older protocol.

gkoh, 2023-07-25:
@93RichardB and @adrianuseless
I upgraded my X-T30 to v2.01 and confirmed the existing implementation still works.
A quick look at the Bluetooth traffic log with the new Xapp suggests the protocol has not been changed.

Can either/both of you please elaborate what exactly no longer works?

NB. I had to delete `furble` from the camera pairing then scan and pair again after updating.

93RichardB, 2023-07-29:
I open blue tooth pairing on the XT4, get message ‘start pairing with compatible device’, press connect on the M5, it shows ‘6228X-T4-6228, click OK,,  goes to thermometer ‘connecting’, then ends and goes back to connect/Scan/Delete… menu.  Camera indicates no connection.

 

Actually, now after doing this a couple of times, selecting connect leads to a menu that just has ‘Back’, no thermometer.

 

Hope this helps.

gkoh, 2023-08-07:
@93RichardB Thank you for the reply!

I would like to confirm the following items:
* previous pairing connections were deleted
   * on both camera and `furble`
* a smartphone is not nearby and/or bluetooth is disabled
   *  with my Android device, the new X app aggressively pairs in the background with no human intervention :neutral_face:
   * this had me really confused for a while
* the target camera shows up when a new pairing request is started (ie. it appears in the `Scan` menu)

gkoh, 2023-08-09:
> Deleted Bluetooth connections on phone and furble. Entered pairing on phone, selected scan on furble.

I assume when you say 'phone' you actually mean camera?

> Scan found 4 identical instances of the XT4 (all same serial), chose one and selected connect.

Woah, that's weird.
I will see if I can reproduce this, that is unexpected.

> It connected! Pressed shutter release button on furble, and furble lost connection and went back to Connect/Scan/Delete menu. No photo taken. Under delete it still shows the XT4 as present, but connecting fails, unless I delete pairing first, then same sequence of events. Hope that helps.

This is progress of sorts, definitely useful.

93RichardB, 2023-08-09:
Oops.  Yes Camera not phone.  I did turn Bluetooth off on my iPhone before I started.

Richard

gkoh, 2023-08-20:
> > Scan found 4 identical instances of the XT4 (all same serial), chose one and selected connect.
> 
> Woah, that's weird. I will see if I can reproduce this, that is unexpected.

@93RichardB
I cannot reproduce this behaviour, it's not supposed to be possible.
The underlying Bluetooth LE library filters out duplicate advertisements during scanning.
Is there a bug here ... or perhaps we are seeing non-duplicate entries ... hmmm.

This leads to further questions:
1. During scanning does the 'Found' message box say 4?
2. Are you able to access the serial port to see the debug output?

If you can access the serial port, I can generate a debug build to see if we can get some more insights.

93RichardB, 2023-08-20:
How could I access the serial port?  Let’s me know how to do it and I will try.

Richard

Richard

gkoh, 2023-08-21:
> How could I access the serial port? Let’s me know how to do it and I will try. Richard Richard Sent from my iPad

Sure! Thanks for helping.
Firstly, what operating system would you be using?

The serial port of the M5StickC is same USB-C cable connection used to program the device, so we just need to get you an app to connect and read from it during a scan & connect.

93RichardB, 2023-08-21:
I can use Windows 10 or Linux (mint).  Most comfortable with Windows.

gkoh, 2023-08-22:
> I can use Windows 10 or Linux (mint).  Most comfortable with Windows.

@93RichardB Excellent, thank you!
OK, let's try the _simplest_ option first, using some online browser based tools, through Windows.
I haven't had great success with this, but ... here goes:
* Go here: https://serial.huhn.me/
* Plug in your M5StickC-Plus
* Click 'Connect'
* Find your device and connect

It will either work and you will see readable output.
Or it will issue an error about being unable to connect (if this happens, we probably need to use a specific app).

Let me know how you go.

gkoh, 2023-08-22:
@93RichardB
You should output like the following during a scan:
```
Name = 043DX-T30-043D
Address = 3c:ff:ff:ff:53:6b
Token = 103450
```

Then something like this during a successful connect:
```
Connecting
Connected
Pairing
Token = 103450
Paired!
Identifying
Identified!
Configuring
Configured
Adding new entry
Saved 3CFFFFFF536B
Index entries: 1
```

gkoh, 2023-08-23:
Also, note that I have created a new discussion (#47) to try and capture more data points from the community.

gkoh, 2023-08-24:
> Result of scan:

@93RichardB Woah, that's excellent, thanks for the output!
I will likely ask for a few more rounds as we debug this, I hope this is OK.

This is totally unexpected, it's a full system crash and reboot.

I think there are 2 problems, separate but probably related:
1. The scan has multiple identical entries
2. It crashes in the "Remote Control" loop

Give me a bit to review the code. I will probably build a debug image for you to test.

93RichardB, 2023-08-24:
Happy to help!

 

More rounds is fine

 

Richard

gkoh, 2023-08-24:
@93RichardB
Please try v1.1.0-debug here:
https://github.com/gkoh/furble/releases/tag/v1.1.0-debug

There are no code changes, but I have rebuilt and added the elf image so I can debug the stacktrace dumped on the serial port.
It will be _very_ interesting if it doesn't reproduce.

gkoh, 2023-08-25:
> @93RichardB Please try v1.1.0-debug here: https://github.com/gkoh/furble/releases/tag/v1.1.0-debug
> 
> There are no code changes, but I have rebuilt and added the elf image so I can debug the stacktrace dumped on the serial port. It will be _very_ interesting if it doesn't reproduce.

@93RichardB 
Just to be clear, about what I need:
* Please reimage your M5StickC with the debug `.bin`
* Please repeat the serial port stuff you did before with the debug image
* Please paste the serial port output here again

With the associated elf image, I should be able to work out where it crashed.

gkoh, 2024-03-13:
No further reports have been received, closing this issue as stale.

closed 2024-03-13 by gkoh

gkoh, 2024-03-14:
> Sorry for not getting back to you. Lent my camera to my son for a few weeks and then didn’t pick back up. Will try again this week.

No problem, please use the pre-release v2.0.0-beta and the web installer to ensure your device is properly programmed.

93RichardB, 2024-09-16:
Sorry for the delay.  Eventually I was unable to test the XT-4 again, but I can confirm that the X-H2 works!

gkoh, 2024-09-17:
> Sorry for the delay. Eventually I was unable to test the XT-4 again, but I can confirm that the X-H2 works! 

No problems @93RichardB, thank you for the confirmation and glad it works for you now (albeit on a different/new camera).

adrianuseless, 2025-01-26:
Will it be able or can it be made to work with the Nikon z range of cameras?

gkoh, 2025-01-27:
> Will it be able or can it be made to work with the Nikon z range of cameras?

This question belongs in the discussion section, not in an old and closed issue.
Please raise a new discussion topic.