gkoh/furble #65: Add shutter lock mode

gkoh, 2024-02-29:
Add a remote control mode that supports shutter lock.
Use case is described by @matthudsonau [here](https://github.com/gkoh/furble/issues/63#issuecomment-1970140462).

matthudsonau, 2024-02-29:
I did implement this in my own branch based on an old codebase with the quick press bug (#63). My knowledge of git is pretty poor/non-existent, and my coding is probably at a similar level, so it's debatable if it'll be of any use

gkoh, 2024-02-29:
Feel free to cut'n'paste your code right into this issue if you wish, happy to consider ideas.

matthudsonau, 2024-02-29:
I apologise in advance for how bad it is; it's been a few years (ok, closer to a decade) since I last coded in C

```
static void remote_control(Furble::Device *device) {
  Serial.println("Remote Control");
  
  int toggle_shutter = 0;
  int toggle_focus = 0;

#ifdef M5STACK_CORE2
  ez.msgBox("Remote Shutter", "", "Release#Focus#Back", false);
#else
  ez.msgBox("Remote Shutter", "\nBack: Power", "Release#Focus", false);
#endif
  while (true) {
    M5.update();

    update_geodata(device);

#ifdef M5STACK_CORE2
    if (M5.BtnC.wasPressed()) {
      break;
    }
#else
    if (M5.BtnPWR.wasClicked()) {
      break;
    }
#endif

    if (M5.BtnA.wasPressed()) {
      if (toggle_shutter == 0){
        toggle_shutter = 3;
        device->shutterPress();
      }
      
      if (toggle_shutter == 2){
        toggle_shutter = 1;
        device->shutterRelease();
      }
      
      toggle_focus = 0;
    }

    if (M5.BtnA.wasReleased()) {
      toggle_shutter = 1;
      toggle_focus = 0;
      
      device->shutterRelease();
    }

    if (M5.BtnB.wasPressed()) {
      if (toggle_focus == 0){
        toggle_focus = 3;
        device->focusPress();
      }
      
      if (toggle_focus == 2){
        toggle_focus = 1;
        device->focusPress();
      }
      
      toggle_shutter = 0;
    }

    if (M5.BtnB.wasReleased()) {
      toggle_focus = 1;
      toggle_shutter = 0;
      device->focusRelease();
    }

    if (toggle_shutter == 1){
      toggle_shutter = 0;
#ifdef M5STACK_CORE2
      ez.msgBox("Remote Shutter", "", "Release#Focus#Back", false);
#else
      ez.msgBox("Remote Shutter", "\nBack: Power", "Release#Focus", false);
#endif
    }

    if (toggle_shutter == 3){
      toggle_shutter = 2;
#ifdef M5STACK_CORE2
      ez.msgBox("Remote Shutter", "Shooting...", "Release#Focus#Back", false);
#else
      ez.msgBox("Remote Shutter", "Shooting...\nBack: Power", "Release#Focus", false);
#endif
    }

    if (toggle_focus == 1){
      toggle_focus = 0;
#ifdef M5STACK_CORE2
      ez.msgBox("Remote Shutter", "", "Release#Focus#Back", false);
#else
      ez.msgBox("Remote Shutter", "\nBack: Power", "Release#Focus", false);
#endif
    }

    if (toggle_focus == 3){
      toggle_focus = 2;
#ifdef M5STACK_CORE2
      ez.msgBox("Remote Shutter", "Focusing...", "Release#Focus#Back", false);
#else
      ez.msgBox("Remote Shutter", "Focusing...\nBack: Power", "Release#Focus", false);
#endif
    }

    ez.yield();
    delay(50);
  }
}
```

gkoh, 2024-02-29:
> I apologise in advance for how bad it is

No problems, I appreciate the effort!

So, reading what you described and your code I tried a few ways to reliably enter shutter lock:
* detect double/triple click
** super unreliable due to the underlying library debounce weirdness
* hold focus + click release, then let go
** this was the most reliable way
** to exit, press either focus or release again

If you are able to program your target, try the branch on this issue.
If not, I can build a test release for you.

Eventually we should also improve the recent interval mode to be easy to configure:
* interval count
* inter-shot delay
* intra-shot shutter time

I'm still thinking on how to do that intuitively with a tiny screen and two buttons.

matthudsonau, 2024-02-29:
Branch working as described. Also fixes the quick tap issue

(side note: you can also add the 100ii onto the supported list of cameras)

I haven't got any good ideas re: interval timer. The best one so far is to have the side buttons go up and down, and use the main button as start, then if you hold the main button you exit the timer mode. That'll make the UI designers cry though, because the buttons then have a unique behavior in just one part of the app

gkoh, 2024-02-29:
> Branch working as described. Also fixes the quick tap issue

Excellent! Thanks for testing.

> (side note: you can also add the 100ii onto the supported list of cameras)

Done.
 
> I haven't got any good ideas re: interval timer. The best one so far is to have the side buttons go up and down, and use the main button as start, then if you hold the main button you exit the timer [...]

Yep, tiny screen + no buttons = UI design nightmare.
I have some random ideas, but they all suck.

matthudsonau, 2024-02-29:
I'm probably going to solve the UI issue by switching to the Core2. Three usable buttons and a touch screen beats the lonely 2 on the m5stick

I might see if I'm clever enough to program my idea in and test. I expect I'll run into problems, but it'll be a good exercise

closed 2024-02-29 by gkoh

gkoh, 2024-02-29:
> I'm probably going to solve the UI issue by switching to the Core2. Three usable buttons and a touch screen beats the lonely 2 on the m5stick

Indeed. The primary target is the M5StickC, so whatever we do needs to be adaptable.
May need to refactor the UI into 2 separate modes, StickC and Core.
 
> I might see if I'm clever enough to program my idea in and test. I expect I'll run into problems, but it'll be a good exercise

Great, lemme know  how you go.
The interval stuff there is a basic placeholder as per #51, but a good base to fiddle.