gkoh/furble #67: Add Advanced Interval Shutter Release

matthudsonau, 2024-02-29:
As started in #51 by @sbstjn and added in #60 , creating a more feature rich interval timer

matthudsonau, 2024-02-29:
Got a test UI up on my own fork (I think). It's...ok, at best. I'm not sure the power button is the most responsive, and I really don't like using a loop+counter for timing. This really should be done with some kind of hardware timer, but it's outside my realm of knowledge (for now)

matthudsonau, 2024-03-01:
Yeah, I found out about the hardcoded bits. No way to bypass that unfortunately (that's I've found yet, at least)

I did manage to get a timer based intervalometer working last night. Only problem is that once it's turned on you can't turn it off, and it'll crash the stick if you exit the intervalometer mode. But at least non loop based timing is something that's possible

gkoh, 2024-03-01:
I've pushed my attempt at a spinner UI for entry of the intervalometer parameters to a branch.
Feel free to test it out.
It currently adds 2 entries right at the top level menu so you can see what it looks like.
I think it's the best I can with 2 buttons.

As of now it supports:
* units of milliseconds, seconds and minutes
* integers up to 999

It does not save the entered numbers (just holds them in some a global variable).

gkoh, 2024-03-01:
> Yeah, I found out about the hardcoded bits. No way to bypass that unfortunately (that's I've found yet, at least)
>
> I did manage to get a timer based intervalometer working last night. Only problem [...]

The primary Arduino loop is based on a hardware timer, so using a loop+counter isn't terrible.
As the timing requirement isn't that strict and we don't need to do anything else I think it's OK.

Having said that, rather than using `delay()`, it's more robust to:
* run a loop with a say, 1-10ms delay, so you can poll the buttons and not burn the MCU into the ground
* use `millis()` to determine elapsed time and do things
  * I did something similar with the shutter lock msgbox counter

matthudsonau, 2024-03-01:
Branch wouldn't compile (extra parenthesis on line 40 in spinner.io). Fixed that and got the UI working

I'm not a massive fan of the UI, but I can't really see any other way to do things with only two buttons. The one thing it'll definitely need in an infinity option for the count

Maybe having some kind of preset value/s, then the fully custom spinner for each input? Just common things like 1s, 2s, 5s, 10s, 30s, 60s and custom?

gkoh, 2024-03-01:
> Branch wouldn't compile (extra parenthesis on line 40 in spinner.io). Fixed that and got the UI working

Ooops, sorry, still hacking the heck out of it.
Thanks for testing it out.

> I'm not a massive fan of the UI, but I can't really see any other way to do things with only two buttons. The one thing it'll definitely need in an infinity option for the count

I hadn't even considered that ... OK difficult to spin up to infinity.

> 
> Maybe having some kind of preset value/s, then the fully custom spinner for each input? Just common things like 1s, 2s, 5s, 10s, 30s, 60s and custom?

Yes, this is also a possibility and would then handle the infinity case.
Now we have some possibilities we can likely do something that is usable.

gkoh, 2024-03-02:
> Maybe having some kind of preset value/s, then the fully custom spinner for each input? Just common things like 1s, 2s, 5s, 10s, 30s, 60s and custom?

The input spinner could have 2 modes, preset and custom, swappable with the 'BtnB + BtnA' hold trick from the shutter lock work.

I think delay and shutter times are the easiest to have preset times, if we use 'standard' shutter speeds, we get:
* 1
* 2
* 4
* 8
* 16
* 32
* 64
* 125
* 250
* 500

The unit spinner can always be present to scale the durations, thus shutter speeds will mostly use millisecs whilst delay would mostly use seconds or minutes.

For the count ... I'm not sure how or which presets would work well.
Perhaps it is simpler to just leave it as the spinner always and treat '000' as infinite.

I will try to get this prototyped over the next few days to see if/how it works.

gkoh, 2024-03-06:
I'm still working on this but got distracted with other things (eg #69).

closed 2024-03-10 by gkoh