gkoh/furble #75: M5StickC Plus range seems lower than M5StickC

gkoh, 2024-03-10:
> Have you tried increasing the transmit power in settings?

Yeah, it was on max. Ended up just dropping the remote below the tripod whenever I needed to go more than a couple of metres away

_Originally posted by @matthudsonau in https://github.com/gkoh/furble/pull/73#discussion_r1518698209_

gkoh, 2024-03-10:
Also this:
> Hmm, interesting. I just tested both M5StickC and M5StickC Plus.
> For M5StickC:
> * connect works at up to ~10 metres
> * shutter release works up to ~12 meters
>   * it can probably go a bit further, I ran out of house and didn't want to go outside
> For M5StickC Plus:
>* connect works at up to ~3 metres
>* shutter release works up to ~10 metres
>* definitely much worse than M5StickC
> That seems weird, I have a M5Stack Core as well, will play around more later.

From https://github.com/gkoh/furble/pull/73#discussion_r1518700543

matthudsonau, 2024-03-10:
Damn, I was hoping it was my fault. Potential hardware limitations aren't fun

My Plus2 should be arriving tomorrow or Tuesday (plus a Stack), so I can test on that once I hack it into working. I don't have an original stickC, so I can't use that as a point of comparison

gkoh, 2024-03-10:
I may have found a fix, I never actually read the ESP32 BLE power control properly, but it says:
> ESP_BLE_PWR_TYPE_CONN_HDL0-8: for each connection, and only be set after connection completed.
> when disconnect, the correspond TX power is not effected.
from [here](https://github.com/espressif/esp-idf/blob/c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b/components/bt/include/esp_bt.h#L210)

I just tried by copying this fragment (from `furble.ino::setup()`):
```
  // Set BLE transmit power
  esp_power_level_t esp_power = settings_load_esp_tx_power();
  NimBLEDevice::setPower(esp_power);
```
and adding it to the start of `menu_remote()`, which is _after_ we establish the BLE connection to the camera.
AND ... it seems to do something, shutter control is now seems on par with my M5StickC.
Connection distance seems unchanged.

I'm in the middle of a huge refactor, but will commit/merge then raise a proper PR.

closed 2024-03-10 by gkoh