gkoh/furble #83: Add per-remote identifier support

gkoh, 2024-03-14:
If you have more than one remote they're all called `furble`.
This makes it difficult/messy if you're developing/testing or doing interesting things with multiple cameras and multiple remotes.

Add a way to identify the remote via name. Perhaps using some fragment of the ESP32 ChipID?

matthudsonau, 2024-03-14:
I think defaulting to something unique is a great idea. Would be nice if it could be changed as well, but something tells me that that's a UI nightmare to do

gkoh, 2024-03-14:
> I think defaulting to something unique is a great idea. Would be nice if it could be changed as well, but something tells me that that's a UI nightmare to do

Agreed, I'm also moving this to 2.0 because it's getting a bit challenging juggling remotes.

closed 2024-03-14 by gkoh