gkoh/furble #86: Backlight doesn't turn off on the M5StickC Plus2

matthudsonau, 2024-03-14:
Instead of the backlight turning off, you get a fun strobe effect. Guess the LCD pinout change wasn't entirely without problems...

gkoh, 2024-03-14:
Hmm, looks like something is broken with the M5Unified library at v0.1.13.
I had a similar flicker problem with my M5Stack Core with 0.1.12, however now I can't even change the brightness :/

 You could try downgrading the library:
* in platformio.ini
* change `M5Unified@0.1.13` -> `M5Unified@0.1.12`

I will investigate further.

gkoh, 2024-03-14:
Erg, it could also be the M5GFX library underneath M5Unified that also got auto-bumped from 0.1.12 to 0.1.13.

gkoh, 2024-03-15:
Well, that's nice, now I can't control backlight brightness for my M5Stack Core at all, with either 0.1.12 or 0.1.13.

gkoh, 2024-03-15:
> Instead of the backlight turning off, you get a fun strobe effect. Guess the LCD pinout change wasn't entirely without problems...

Looks like they also ripped out the PMIC, so backlight control is now via PWM.
M5StickC and Plus use the AXP192, brightness control seems fine there, but Plus2 and Core use PWM, which appears to not work or at least work wrong.

The full documented differences according to M5Stack are here:
* https://docs.m5stack.com/en/core/M5StickC%20PLUS2

Scroll down to "The difference between M5StickC PLUS and M5StickC PLUS2".

gkoh, 2024-03-15:
Looks like the PWM controls for M5Stack Core aren't being called _at all_.
My M5StickC Plus2 should arrive in the next week or so, when that happens I will check there.
This one might take a bit of digging about in the underlying libraries to fix.

gkoh, 2024-03-15:
Gah, I cut a minimal sketch to test and I can control the backlight properly. I will continue to narrow down the issue.

gkoh, 2024-03-15:
> Gah, I cut a minimal sketch to test and I can control the backlight properly. I will continue to narrow down the issue.

Found it, there are 2 problems:
* The serial port pins used to support GPS are different on the M5Stack and interfere with the backlight pins :(
   * I will need to parameterise them and test when my M5Stack Core battery comes
* PWM control doesn't work during `lightSleep()` by default
   * I need to dig into platformio configuration to find out if we enable it

gkoh, 2024-03-21:
My M5StickC Plus2 arrived and I have fixed this.
On this and other PWM backlight devices we avoid `lightSleep()` as this disables the PWM and hence the backlight.
To try to conserve power, lower the CPU to 10MHz.
However, this messes up the clocks and the serial ports get clocked at the wrong frequency.

For now, accept this as 'idle' isn't where the fun happens.

closed 2024-03-21 by gkoh