gkoh/furble #95: GFX100II GPS Not Working

gkoh, 2024-06-21:
### Discussed in https://github.com/gkoh/furble/discussions/94

<div type='discussions-op-text'>

<sup>Originally posted by **edcooley** June 20, 2024</sup>
New user installed on Plus 2 with web install.  I can connect to my GFX100 II and fire the shutter fine.  I am having issues with GPS not recording on exposures.  The GPS option is turned on and GPS data shows date and location.  The option to request smartphone location is enabled on the body and using the Xapp to fire the shutter adds GPS to images.

Any idea what I am doing wrong?</div>

gkoh, 2024-06-21:
@edcooley Hmm, I thought I had it reproduced, not so sure now.
If possible, let's grab some logs to confirm the basics are OK:
* connect your Plus2 to your computer
* go to https://furble-web-installer.koh.wtf/
* `Connect` to the Plus2
* Select "Logs & Console"
   * this brings up the serial console of the Plus2 (and furble's log output)
* On the Plus2:
   * Connect to the camera
   * Enter Shutter (furble only updates GPS when in shutter control, this may need to change)
 * Back in the browser, you should see output

NB. Please redact your actual location data before posting here.

On my X-T30 I see this output:
```
Remote Control
Got notification callback: 2 bytes from 0x5042
  [0] 0x01
  [1] 0x00
Sending geotag data (23 bytes) to 0x8012
  lat: a.b, ab
  lon: c.d, cd
  alt: e.f, e
Got notification callback: 2 bytes from 0x5042
  [0] 0x01
  [1] 0x00
Sending geotag data (23 bytes) to 0x8012
  lat: a.b, ab
  lon: c.d, cd
  alt: e.f, e
Exit shutter
```

"Got notification callback ... from 0x5042" means the camera has requested a geotag update.
"Sending geotag data ... to 0x8012" means we are sending the following GPS data.

If you're seeing something similar then the furble side of things is working.

edcooley, 2024-06-21:
Not seeing any of that.  It does fire the shutter but getting unhandled exceptions every few seconds.

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[   228][E][Preferences.cpp:483] getString(): nvs_get_str len fail: theme NOT_FOUND
Index entries: 1
Loading index entry: 387C76C1CD1F
Index entries: 1
Loading index entry: 387C76C1CD1F
Connecting
Connected
Pairing
Token = 175d2080
Paired!
Identifying
Identified!
Configuring
Got notification callback: 2 bytes from 0x5009
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
Configured
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
Remote Control
shutterPress()
shutterRelease()
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00

```

edcooley, 2024-06-21:
Also noticed the focus is flashing the lcd but not actually focusing.

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[   228][E][Preferences.cpp:483] getString(): nvs_get_str len fail: theme NOT_FOUND
Index entries: 1
Loading index entry: 387C76C1CD1F
Index entries: 1
Loading index entry: 387C76C1CD1F
Connecting
Index entries: 1
Loading index entry: 387C76C1CD1F
Index entries: 1
Loading index entry: 387C76C1CD1F
Connecting
Connected
Pairing
Token = 175d2080
Paired!
Identifying
Identified!
Configuring
Got notification callback: 2 bytes from 0x5009
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
Configured
Remote Control
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
focusPress()
focusRelease()
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.
focusPress()
focusRelease()
Got notification callback: 2 bytes from 0x500c
  [0] 0x01
  [1] 0x00
Unhandled notification handle.

```

gkoh, 2024-06-21:
> Not seeing any of that. It does fire the shutter but getting unhandled exceptions every few seconds.

Fantastic work, thanks!

> Also noticed the focus is flashing the lcd but not actually focusing.

Please open a new issue to track the focus, it's a completely undocumented accidental find on my part, Fujifilm may have changed that behaviour. This may not be fixable :slightly_frowning_face: 

As for this issue:
* hmm, we have notifications coming from the camera for 0x500c and 0x5009, I have never seen these before with my old X-T30, let's hope they're not important ...
* I have a theory:
   * in the original camera remote app, there were no controls for geotagging, it seemed to be purely driven by the camera, hence the incoming '0x5042, please send geotag' request
   * in the new X app, there's an option to control location update interval
   * I'm guessing the decision of when to send geotag updates has now been reversed

I will add a new setting to mimic the 'location update interval' setting and mash it into how the previous system works.

gkoh, 2024-06-21:
@edcooley Please try a test version of v3.0.0 from here:
https://4bd0ec87.furble-web-installer.pages.dev/

gkoh, 2024-06-21:
> @edcooley Please try a test version of v3.0.0 from here: https://4bd0ec87.furble-web-installer.pages.dev/

Actually, v3.0.0 is a mistake, the next version will be v2.1.0, not that it's very important, they're just random numbers.

edcooley, 2024-06-21:
Updated version fixed the GPS issue.  

Thanks so much.  This device is huge with the GFX100II as they moved the shutter release socket to the right side of the camera which is inconvenient when using an L bracket.

gkoh, 2024-06-22:
> Updated version fixed the GPS issue.

Excellent, now I need to consider how/whether this works well with older cameras.

> Thanks so much.

You are most welcome.

gkoh, 2024-06-27:
I'm starting to suspect the notifications to `0x5009` and `0x500c` are the new protocol for GPS support.
My best guess from looking at the trace from @edcooley is:
* 0x5009: used by the camera to enable geotagging
* 0x500c: used by the camera to request geotag data

What I don't understand is the new Xapp 'Location Interval' setting and how this all works together.
Without a newer Fujifilm to capture the raw traffic I will be basically playing a guessing game.

edcooley, 2024-06-27:
Happy to record a trace if you point me in the right direction.

gkoh, 2024-06-27:
> Happy to record a trace if you point me in the right direction.

Thanks. Always happy to try!
First requirement, what are you chances you have an Android device to run Xapp?

edcooley, 2024-06-27:
OK, I have a Galaxy tablet with Xapp installed.

gkoh, 2024-06-28:
> OK, I have a Galaxy tablet with Xapp installed. Not that familiar with android so be gentle. 😉

Excellent, I will try to guide you through the following, and hopefully we are successful.
If we are, I will likely turn this into a wiki entry as others have expressed an interest in how to do this.

First up, we should clarify the two scenarios we wish to capture using Xapp:

1. Geotag on camera enabled, connect with xapp, issue a shutter release
2. Geotag on camera disabled, connect with xapp, issue a shutter release

The two scenarios should allow us to isolate/eliminate traffic related to geotagging.
The shutter release acts as a marker, so we know we're looking at the 'right' stream of traffic.

The following has been tested on my Pixel 8 Android phone, this may differ slightly (or a lot) for Samsung (or other brands).

* Enable developer options
   * May differ on different devices, but is generally:
      * Settings->System->About Phone (or About Device)
      * Software Information
      * Tap 'Build number' seven times
* Settings->System->Developer Options
   * Enable Bluetooth HCI snoop log
      * Enabled
* Disable, then re-enable Bluetooth
   * This will restart the bluetooth service with developer log enabled

Now we go through two scenarios above:
* Open Xapp
   * Connect to camera
   * Remote
   * Shutter release
   * Close Xapp
* Change camera to disable geotag
   * Repeat steps above
* Close Xapp

Now we generate a bug report, which will include the bluetooth log:
* Settings->System->Developer Options
   * Bug report
   * Interactive Report
   * 'Report'

This will take a while (few minutes to generate). You will then have the option to 'share' the bug report, choose a convenient option (I tend to just upload to drive).
The file usually measures in the 10's of MBs. I tend to get ~20MB bug reports.

Now disable the HCI snooping:
* Settings->System->Developer Options
   * Enable Bluetooth HCI snoop log
      * Disabled
* Settings->System->Developer Options
   * "Use Developer options" -> Off

Heed the warning:
* the bug report can contain sensitive information, _do not_ share with untrusted parties
   * especially random weirdos that run niche github projects

The bug report is a single zip file, within which we only want one, the HCI snoop log.
On a computer, access the zip file and extract this file, the location can change depending on Android version, for me it was here:
*   `FS/data/misc/bluetooth/logs/btsnoop_hci.log`

This is the file I need, please be aware of the following privacy issues:
* it will contain all bluetooth traffic to/from the Android device
   * it can include any and all traffic from bluetooth peripherals connected to the Android device, including:
      * sensors
      * smart health devices
      * vehicles
      * etc
* if you have any concerns about its contents, _please_ inspect the traffic with a tool like `wireshark`

I absolutely respect your right to privacy and anonymity, as we are dealing with GPS data, the traffic will reveal your exact location.
It is definitely possible to redact all this information before sending it, however it requires time investment.

Please _do not_ attach or upload anything to this project. I will provide you with my email address if you wish to continue.
Please do not send me anything if you are in any way uncomfortable, we can always find other ways to do this without potentially violating your privacy.

edcooley, 2024-07-01:
Please send me your email for the file.Ed Cooley479-381-8307edcooleyfineart.comsteamartandgraphics.com

gkoh, 2024-07-01:
> Please send me your email for the file.

guorong.koh (at) gmail.com

Thanks @edcooley!

gkoh, 2024-07-06:
@edcooley Thanks for the bluetooth snoop log, extremely helpful!
I believe I found the problem, which is how bluetooth services are handled.
Essentially, for the geotag traffic, I (lazily) decided to use the 16-bit short form handle (as opposed to the full 128-bit UUID).
I incorrectly assumed the 16-bit handle was consistent across devices and firmware, it is not, only the 128-bit UUID is, so I have updated the code on the PR.

Please have a try on the deployment branch:
https://f6c7aa00.furble-web-installer.pages.dev/

edcooley, 2024-07-07:
Glad it helped.  Traveling now but will give it a try the end of the week.

https://f6c7aa00.furble-web-installer.pages.dev/

gkoh, 2024-07-18:
I have committed the changes, I'm reasonably convinced it is correct.
At a minimum it doesn't regress on my X-T30.
Please reopen if this does not actually fix it.

closed 2024-07-18 by gkoh