gkoh/furble #111: Backlight timeout brightness is hardcoded and too high

gkoh, 2024-08-11:
Thanks to #108 I reviewed the backlight timeout brightness handling ... and it sucks with multiple problems.

Let's start with the first problem:
* the hardcoded value (as seen in #108) is too high at `64`

If we look at the `ezBacklight::menu()`, the brightness slider goes down to `48`, something I recall determining via trial and error on the original M5StickC.

This means if we set minimum brightness at `48`, we actually _increase_ brightness to `64` during inactivity timeout, this is so wrong.

Second problem:
* the minimum brightness is different for different devices
   * `64` is quite bright for the M5StickC Plus2 and M5Stack Core

I propose the following solutions:
1. Add a private method to `ezBacklight`, say `ezBacklight::getMinimumBrightness()`
   * this should return the minimum viable brightness per device, I experimentally determined the following (subject to input):
      * M5StickC, M5StickC Plus == 48
      * M5StickC Plus2, M5Stack Core[2] == 16
 2. Use `ezBacklight::getMinimumBrightness()` in:
    * `ezBacklight::loop()` during inactivity
    * `ezBacklight::menu()` as the minimum brightness in the slider

gkoh, 2024-08-11:
@hijae Would you be interested in working on this?

hijae, 2024-08-11:
Yes. i want to do. 
And #112 too.

hijae, 2024-08-11:
I noticed while working that even with the backlight set to 0, I can see that the screen is very dimly lit. I don't reflect this because I don't think it would be visible in the sun during the day, but I think it could be lowered if I think the lowest brightness is also bright.

gkoh, 2024-08-12:
> I noticed while working that even with the backlight set to 0, I can see that the screen is very dimly lit. I guess it would be possible to set the backlight to 0 and turn on the power LED.

Possibly, let's implement and then we can adjust as needed.

gkoh, 2024-08-12:
> Yes. i want to do. And #112 too.

OK, I will assign both to you, thank you.

closed 2024-08-14 by gkoh