gkoh/furble #120: Screen Size

ZeFish, 2024-08-26:
I'll be honnest, I didn't take the time to look at the code, but here's my display. Evertything else works except the screen that seems on the smaller size.
[image]

gkoh, 2024-08-27:
@ZeFish That looks like a M5StickC Plus2 (based on the case colour).
I'm testing with `v2.1.0+1`.

If I flash the M5StickC firmware, I get exactly what you see.
If I flash the M5StickC Plus firmware (2nd option in the web installer), I get the right screen resolution.

Can you confirm you are using the correct firmware?

gkoh, 2024-08-27:
Assuming incorrect firmware, this should be prevented in future with #123.

ZeFish, 2024-08-28:
You are right, I originally install the firmware from the github source code, didn't think much of it. Now from the web installer it works. Thanks for taking the time!

closed 2024-08-28 by ZeFish

gkoh, 2024-08-28:
> You are right, I originally install the firmware from the github source code, didn't think much of it. Now from the web installer it works. Thanks for taking the time!

No problem, glad you worked it out (and it's all working for you).