gkoh/furble #128: Add simultaneous multiple camera connection

gkoh, 2024-09-16:
With the shift to expose the EPS32's underlying FreeRTOS, we can further abstract `furble`'s components in meaningful ways.
By adding per-camera tasks and funnelling UI inputs through a command queue, it will be possible to connect to and control multiple cameras at the same time.

To preserve the current 'easy' 'click-click-click' experience, hide the feature behind a setting and default to false.

matthudsonau, 2024-09-16:
Had someone ask about doing this last month. Good timing

gkoh, 2024-09-16:
> Had someone ask about doing this last month. Good timing

Ha!
All the refactoring over the past few months has been aimed at implementing this :slightly_smiling_face: 

Will be great to get real-world feedback on it when it's done.

matthudsonau, 2024-09-16:
Well, once work gets done with firing us (long story) I'll hopefully have a bit of time to test things

gkoh, 2024-09-17:
> Well, once work gets done with firing us (long story) I'll hopefully have a bit of time to test things

:open_mouth:

closed 2024-09-18 by gkoh

gkoh, 2024-09-18:
A pre-release with this feature is in [2.2.0-rc2](https://github.com/gkoh/furble/releases/tag/v2.2.0-rc2)