gkoh/furble #129: Connection without timeout

hijae, 2024-09-18:
I'm using the camera primarily for GPS logging rather than shutter. I would like the camera to continue to wait for a connection even if it briefly sleeps or is powered off, so that I can reconnect at any time.

gkoh, 2024-09-19:
I see, using `furble` as a passive, always on GPS data source.

Adding such a mode will not be kind to battery life, but I assume that is known penalty.
This is certainly possible.

We would need to:
* add a setting to control disconnect behaviour
   * try once, twice, indefinite, etc
   * default to current behaviour
* update the disconnect handler to respect the setting
* add cancel to the (dis)connect handler
   * this one is a bit messy, but nice to have
   * right now there is no way to interrupt or cancel the connect

I'll need some time to think about how to implement the cancel properly.

gkoh, 2024-09-24:
@hijae It appears that properly implementing this reasonably simple request will exceed the capabilities of the underlying M5ez library.
We have been consistently hitting the limits of this ageing library.
I am now exploring alternative UI libraries that can also support #80.

For now, you could hack the UI disconnect handler.
There is a `reconnected` variable which you can just delete. This will send `furble` into infinite reconnection mode.
The only way to exit will be to either:
* connect to a camera then exit properly
* forceful power off by holding the power button

gkoh, 2024-10-03:
I recently worked out this is possible with M5ez, but required a fair amount of rework, see #133.
Moving to an earlier milestone, that doesn't require a new UI.

closed 2024-10-06 by gkoh

gkoh, 2024-10-06:
With this feature, you need to connect successfully once.
After that, the reconnection logic handles things.