gkoh/furble #134: Fujifilm XS-20 doesnt work

kelvincabaldo07, 2024-10-03:
I am trying to connect to my Fujifilm XS-20 to no avail. It can see it when I scan but when I click on it, it doesnt work.
Here is the log
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/17241040/esp-web-tools-logs.txt)

gkoh, 2024-10-04:
@kelvincabaldo07 It's always great to see an issue report with a log, so appreciated :slightly_smiling_face: 

So, summary:
* It's an M5StickC Plus2
* Hmm, we don't log the firmware version, I should do that
   * Can you confirm you are testing with v2.2.0?
* There are 2 errors in the log, as follows ...

```
[ 54407][I][Fujifilm.cpp:144] connect(): [furble] Connecting
E NimBLEClient: Connection failed; status=574 Connection Failed to be Established
```
This is a new one, haven't seen this before (574 = 0x23e)
From other reports related to the NimBLE-Arduino stack, it suggests problems with Bluetooth itself, either environment or hardware.

```
[ 82076][I][Fujifilm.cpp:148] connect(): [furble] Connected
[ 82375][I][Fujifilm.cpp:154] connect(): [furble] Pairing
[ 82562][I][Fujifilm.cpp:60] print_token(): [furble] Token = 00842380
E NimBLERemoteCharacteristic: << writeValue, rc: 7
```

According to https://mynewt.apache.org/latest/network/ble_hs/ble_hs_return_codes.html#return-codes-hci, error code 7  is `BLE_HS_ENOTCONN` == "No open connection with the specified handle."
This strongly suggests that whilst it succeeded in connecting initially, the connection was dropped before `furble` could write the pairing token.

Thus, there is clearly something wrong with establishing a stable BLE connection.
I can imagine some possible scenarios:
* remote is too far from camera
* lots of BLE traffic in the area degrading the SNR
* problem/fault with the M5StickC antenna/BLE hardware
* some bug in `furble` or the software stack that has recently surfaced

@kelvincabaldo07 Are you in a position to help investigate further?

gkoh, 2024-10-04:
Ick, I was just testing the changes for logging and I have this problem now, yay?

kelvincabaldo07, 2024-10-04:
Hi @gkoh , i actually did some test and digging last night and what i did was tried to install an old version by scouring through the old issues. Haha
Luckily i found a link to install an old version and i did that after multiple times of failing on a fresh install of the current version and it worked!
Its not an issue of my proximity as i have it within arms length if not an inch away. Haha.
Firmware is the latest as I cannot select when installing but i looked at the fw in the M5 stick and it is the latest. 
its not hardware as well since it is now working.
I think it is software.
Here’s what i did to make it work.
I used a link to install an old version. That worked. 
then installed the new version without erasing the fw in the device.

Suggestions/Ideas
- provide the ability to select release/version to  install on the web installer
- check the old version vs the new one to see why the issue with nvs is happening

thanks for replying so quickly. I am loving this! 😊

kelvincabaldo07, 2024-10-04:
Once you figure out why. You can close this issue and add Fujifilm XS-20 on the list of compatible devices. Wohoo!🎉 Thanks again for this. Let me know if there’s more I can do to help troubleshooting.

gkoh, 2024-10-04:
@kelvincabaldo07 Good work figuring it out and getting it working.
Yes, good suggestions with the versions, I will need to figure out an elegant solution to that one.

I strongly suspect the upgrade to the underlying BLE stack is causing the problems, watch this space whilst I hack about.

gkoh, 2024-10-04:
OK, I'm wrong and this is super weird and I'm still looking.
This only appears to affect new scan and connect to Fujifilm cameras (which kinda is the whole point of this project :neutral_face:)

If I scan, connect and save a connection from v2.1.0, I can upgrade to v2.2.0 and connect successfully.
With v2.2.0, I cannot scan and connect, it fails with `rc=7` as reported above.

What's weird is the 'new' connection and the 'saved' connection process is identical, so now I suspect something is broken with new connection metadata.

closed 2024-10-04 by gkoh

gkoh, 2024-10-04:
OK, I found it, ended up having to bisect the commits to find the culprit.
I don't understand why it broke, and will investigate further offline.

v2.2.1 has been released to address this, and I tested it this time :slightly_smiling_face: