gkoh/furble #140: Add possibility of changing button config on settings

kelvincabaldo07, 2024-10-14:
Is it possible to add on the settings config on changing how the buttons work?
I was hoping on having the main button be both focus and shutter release. Possibly a single click will be focus and double click or long press would be release or take the shot. 
the current implementation of the focus on B button is a bit harder to use for me so im hoping it is possible. 😊

gkoh, 2024-10-15:
Sorry, long post ahead.

Anything is possible, the current implementation addresses two items:
* separate hardware button for focus and release
   * ala back-AF + shutter
* hold and release

This configuration appears to be acceptable to most users, as both focus and shutter times are independently managed (due to independent buttons), the user is free to do whatever, for me especially:
* focus is sometimes quite slow and requires holding focus longer than expected
* burst shutter, I can hold down shutter for as long/short as I want for CL/CH

I interpret this issue as "closer to the camera experience", where a single finger/thumb can be used to control focus and shutter. Without a double action switch, we cannot replicate this experience exactly.

So, as you suggest, we can approximate it with single and double click (I can't imagine alternatives with a single button, maybe someone else can?):
* single click + long press == focus + hold
* double click + long press == shutter + hold

There are two problems:
* 'double click' is a finnicky thing to define (and capture).
   * my double click is faster/slower than another's
      * I experimented with this early on, it wasn't great
* the current shutter loop is tightly bound to input handling
   * changing to support multiple modes would be yuck

Good news:
* I am currently trialling a UI rewrite as per #80
   * this rewrite separates the input handling from the shutter/camera control
   * this should make trialling different input control configurations much, much easier

tl;dr
Change is coming that should be easier to test and support a request like this.

gkoh, 2025-01-16:
An upcoming version of the UI library, LVGL, is adding double click and triple click event detection.
I will wait for this to be released rather than implementing my own (I tried for v3.0.0 but it got too messy for my liking).