gkoh/furble #150: UI issues in v3.0.0-rc2

matthudsonau, 2025-01-02:
### M5StickC Plus2
- Difficult to tell if the back button (<) is selected in a lot of menus (e.g. settings, scan)
- Back button (<) is displayed but doesn't work on the connected menu
- Unable to lock the shutter in Shutter menu
- Settings->backlight - difficult to tell when the timeout is selected
- Theme select - unclear when the theme is changed or what one is currently selected. Should probably auto-restart if the theme is changed
- Transmit power screen is broken
- Connecting screen should probably remove the bottom button bar
- Scanning only seems to work if you start it on the camera first before scanning on the Plus2

### M5Stack Core2
- I have fat fingers, so hitting the right menu item is a challenge at times
- Shutter lock doesn't work
- Back button on the connected screen doesn't work (probably should be removed as Disconnect is the correct option)
-  Settings->backlight -  unclear how to adjust the timeout. Maybe horizontal instead of vertical scroll with arrows to infer how to use?
- Scanning only seems to work if you start it on the camera first before scanning on the Core2

gkoh, 2025-01-03:
Thanks for this, I will see what can be fixed.

> * Scanning only seems to work if you start it on the camera first before scanning on the Core2

Ha, I thought this was just my imagination. Thanks for detailing the reproduction steps.

> * Back button (<) is displayed but doesn't work on the connected menu

I struggled with this. It's actually disabled (greyed out) to force you to use `Disconnect`.
The menu widget didn't like me removing it (kept putting it back! :neutral_face:).
I guess there are a few options:
* try harder to remove it
   * theoretically I could write it from scratch, I'd rather not
* leave it enabled but:
   * rename `Back` to `Disconnect`
   * makes the StickC version (with no label unobvious)

> * I have fat fingers, so hitting the right menu item is a challenge at times

I have this problem as well (I keep hitting `Power Off` when I actually want `Settings`).
For now, we can bump the font size up a little.
Ultimately, I think the right solution for touch screens is to convert to icons in a grid?
This would be a future bunch of work.

gkoh, 2025-01-03:
@matthudsonau Separately, could you please email me at `guorong.koh@gmail.com` I have a question I'd like to ask offline. Thanks.

matthudsonau, 2025-01-03:
> Ha, I thought this was just my imagination. Thanks for detailing the reproduction steps.

Turns out when you're connecting multiple remotes for testing you sometimes run into problems. I had other problems as well, but I haven't been able to reproduce them on rc2 so I figure they're already fixed

> I struggled with this. It's actually disabled (greyed out) to force you to use `Disconnect`. The menu widget didn't like me removing it (kept putting it back! 😐). I guess there are a few options:

Could you set the colour to be the same as the background? That was it's still there, just invisible?

> I have this problem as well (I keep hitting `Power Off` when I actually want `Settings`). For now, we can bump the font size up a little. Ultimately, I think the right solution for touch screens is to convert to icons in a grid? This would be a future bunch of work.

I think that's the better option (of course it is, because it's more work). I actually prefer the Core2's interface now, the touchscreen is a lot easier to navigate with

matthudsonau, 2025-01-03:
One other thing that might be worth investigating with the Core2 is some way to disable the touchscreen using the physical buttons. I know there's times when I'm out on location where I drop the remote in a pocket, so preventing erroneous touches would be useful

gkoh, 2025-01-03:
> > Ha, I thought this was just my imagination. Thanks for detailing the reproduction steps.
> 
> Turns out when you're connecting multiple remotes for testing you sometimes run into problems. I had other problems as well, but I haven't been able to reproduce them on rc2 so I figure they're already fixed

I just tried with rc2 and came to the same conclusion.
I couldn't reproduce (and I didn't do anything magic in rc2 ... I'll keep on eye on it).

> > I struggled with this. It's actually disabled (greyed out) to force you to use `Disconnect`. The menu widget didn't like me removing it (kept putting it back! 😐). I guess there are a few options:
> 
> Could you set the colour to be the same as the background? That was it's still there, just invisible

Ah, I will add that to the list of solutions, thanks.

> I think that's the better option (of course it is, because it's more work). I actually prefer the Core2's interface now, the touchscreen is a lot easier to navigate with

:+1:

gkoh, 2025-01-03:
> One other thing that might be worth investigating with the Core2 is some way to disable the touchscreen using the physical buttons. I know there's times when I'm out on location where I drop the remote in a pocket, so preventing erroneous touches would be useful

Oh, like a 'lock' button. I'll create a new issue to track this.

gkoh, 2025-01-03:
> # M5Stack Core2 
> * Shutter lock doesn't work

I tried to be smart, it's a long press to activate and deactivate, as a 'deliberate' action.
Can be changed to normal click on/off.

Hmm, could even be:
* switch toggle (eg. left/right)
* checkbox

matthudsonau, 2025-01-03:
> I tried to be smart, it's a long press to activate and deactivate, as a 'deliberate' action. Can be changed to normal click on/off.

Hey, that works. Fine with that behaviour, just needs to be indicated somewhere to avoid this situation

> Hmm, could even be:
> 
> * switch toggle (eg. left/right)
> * checkbox

If I'm allowed to dream big, mirroring the app (press and slide) might be a good idea. Likely difficult to implement though

gkoh, 2025-01-04:
> > I tried to be smart, it's a long press to activate and deactivate, as a 'deliberate' action. Can be changed to normal click on/off.
> 
> Hey, that works. Fine with that behaviour, just needs to be indicated somewhere to avoid this situation

I documented it in the README, but you have to wade through oodles of text to find it.

> > Hmm, could even be:
> > 
> > * switch toggle (eg. left/right)
> > * checkbox
> 
> If I'm allowed to dream big, mirroring the app (press and slide) might be a good idea. Likely difficult to implement though

I tried this early on, but failed as I wasn't familiar enough with the graphics library.
However, I think this is possible without too much head-banging.
I'm going to create another issue to improve touch screen experience:
* prefer icons over menu
* add shutter lock press and slide

I've tried to address all the items identified and will release rc3 soon.

closed 2025-01-04 by gkoh

gkoh, 2025-01-04:
3.0.0-rc3 is now released with attempts to address all items mentioned.
If new issues or more is needed, please reopen this or raise a new issue, much thanks.

matthudsonau, 2025-01-04:
> 3.0.0-rc3 is now released with attempts to address all items mentioned. If new issues or more is needed, please reopen this or raise a new issue, much thanks.

No worries, I'll do some testing (if I'm lucky that'll be field testing) and open a new issue with anything I find