gkoh/furble #151: Improve battery during Infinite-ReConnect

gkoh, 2025-01-03:
@hijae I had a thought with improving battery life when using `Infinite-ReConnect`.
The default connect timeout is 30s, this means the device will continuously attempt connect for 30s, thus the duty cycle is effectively 100%.

What if, during `Infinite-Reconnect`, we reduce the connect timeout to, say, 10s.
And sleep for, say, 10s, this would reduce the duty cycle to 50%.
This should improve lifetime somewhat (although GPS looks like quite a large power sink).

We could adjust the numbers to ensure responsive to the camera going active (ie, you don't want to wait 60s for the next connect attempt).

hijae, 2025-01-16:
I think it's a great idea, because I'm using 18650 HAT(M5StickC 18650C) to get more battery life.

hijae, 2025-01-17:
I've used it for about 12 hours straight, but if you need a longer test, I'll do it.

gkoh, 2025-01-17:
> I've used it for about 12 hours straight, but if you need a longer test, I'll do it.

Not necessary, but thank you for offering.
I was interested in whether the current draw estimates were accurate or even realistic.

I will work on this change, 10 seconds connect, 10 seconds sleep, during multi-connect.