gkoh/furble #159: Fujifilm scan not reliable

gkoh, 2025-01-07:
With v3.0.0-rc4 I can semi-reproduce the scan issue.

If `Scan` is enabled, then left for 'some time', a Fujifilm camera that is then set to pair does not appear.

From the debug, the NimBLE stack definitely picks up the new advertising device, but for some reason the callback to add it to the menu doesn't happen.

Need to investigate further.

FYI @matthudsonau

gkoh, 2025-01-07:
I've found one way to reproduce:
* pair camera
* delete from furble
* start scan
* start camera
   * realise old pair is still on camera
   * delete from camera
* start pairing
* doesn't show up

This happens because the camera starts and immediately tries to connect to furble, but the scan data doesn't include the pair information. As the scan filter drops duplicates based on Bluetooth MAC address, it never shows again.

gkoh, 2025-01-07:
Hmm, I cannot reproduce this any other way.
Leaving this here to keep an eye on it.

Grumpytech4, 2025-07-11:
I can also see the same problem where the scan is started first and then the camera placed in pairing mode that the Furble will not detect the camera. If the scan is cancelled and the camera placed in pairing mode first then the scan started the furble will detect the camera.

gkoh, 2025-07-16:
> I can also see the same problem where the scan is started first and then the camera placed in pairing mode that the Furble will not detect the camera. If the scan is cancelled and the camera placed in pairing mode first then the scan started the furble will detect the camera.

Was the camera previously paired?