gkoh/furble #167: On M5StickC portrait orientation, button icons no longer align with actual buttons

gkoh, 2025-01-16:
It's great works. But In m5stack, the buttons are not intuitive as the screen changes from landscape to portrait. The right button is at the top, so it would be nice to move it up.

_Originally posted by @hijae in https://github.com/gkoh/furble/discussions/164#discussioncomment-11849883_

gkoh, 2025-01-16:
I had plans to do this and admin I completely forgot with the rest of the v3.0.0 changes.

I recall when I first tried, the button icon was covering some menu text in some cases, so I moved it out of the way.
Let me experiment with it a little.

@hijae If you want to try yourself, it's quite straightforward.
If you look here:
https://github.com/gkoh/furble/blob/649c17d2c55714ba0314ec7556b899546b4f8ab1/src/FurbleUI.cpp#L263

and change `LV_ALIGN_BOTTOM_RIGHT` to `LV_ALIGN_RIGHT_MID`, it should put it in the 'right' place.

The alignment values can be found here:
https://docs.lvgl.io/9.2/overview/coord.html#align

gkoh, 2025-01-17:
[image]

@hijae How's this?
(sorry for the bad photo)

hijae, 2025-01-17:
I think no problems with the UI. But if I press the right button, it moves down.
but right button is on top.
So I think switch left and right, or change icon from left and right to up and down is batter choose.

[image]

gkoh, 2025-01-17:
Aha, I misunderstood your original comment. Thank you for the image, now I understand and agree.

In landscape, the 'B' button was always 'down/next', so I will keep that, otherwise existing users will need to relearn.
I will modify the symbols.

closed 2025-01-18 by gkoh

gkoh, 2025-01-18:
OK, it looks like this now:

[image]

The arrows are the default built-in symbols, which I don't really like, but suffice.
I'm waiting on an update to the UI library (lvgl) to properly support rendering SVG so I can custom draw some.

gkoh, 2025-01-18:
I should also note:
* when navigating menus, the buttons show 'up/down'
* when in a slider (eg. brightness, transit power), the buttons change to 'left/right'