gkoh/furble #181: Connect to multiple cameras at once?

cpkcpkcpk, 2025-04-13:
Hey,

Probably a long shot, but just wondering if it would be possible to trigger 2 Fujifilm X-S20s together from one Furble? 

Otherwise could you sync 2 Furbles connected to separate cameras, so you would only have to press the shutter on one?

Thanks!

cpkcpkcpk, 2025-04-13:
Sorry I totally missed the multi-connect section of the readme, so happy to see that! I imagine it would be impossible to get frame-accurate sync over bluetooth, but will be interesting to see how close it is.

gkoh, 2025-04-14:
> Sorry I totally missed the multi-connect section of the readme, so happy to see that! I imagine it would be impossible to get frame-accurate sync over bluetooth, but will be interesting to see how close it is.

@cpkcpkcpk Very glad you managed to find the multi-connect feature. It certainly isn't well publicised and isn't quite as well tested as I would like.
Unfortunately, I don't have enough hardware to do proper testing.

You are correct, bluetooth latency will never compete with a hard wired solution.
Please test and report back if you are able.
If the lag is too large, there are some minor tweaks I can make which may improve multi-connect somewhat.

For now, I will close this issue, feel free to re-open or open a new one if 'sync lag' is problematic.

cpkcpkcpk, 2025-04-24:
Hi, I've been testing this with an X-S20 and X-S10 and it works reasonably well at 750ms interval, but there is quite a bit of lag which can be seen in the attached video. If I go down to 500ms then it starts dropping some frames. Both cameras have a 3fps slow burst mode, so presumably they should be able to handle 333ms until the buffer is full.

I'm not sure if the lag is due to bluetooth latency, or if the older processor in the X-S10 is just struggling to keep up with the newer X-S20. 

It would be interesting to know if this could be improved at all with some tweaks! 

https://github.com/user-attachments/assets/08304648-457e-4688-8148-10265b592e05

gkoh, 2025-04-25:
> It would be interesting to know if this could be improved at all with some tweaks!

Thank you for continuing to test this!
I've spun this out to #183 and we continue there.