gkoh/furble #183: Multi-Connect lag issues

gkoh, 2025-04-25:
> Hi, I've been testing this with an X-S20 and X-S10 and it works reasonably well at 750ms interval, but there is quite a bit of lag which can be seen in the attached video. If I go down to 500ms then it starts dropping some frames. Both cameras have a 3fps slow burst mode, so presumably they should be able to handle 333ms until the buffer is full.
> 
> I'm not sure if the lag is due to bluetooth latency, or if the older processor in the X-S10 is just struggling to keep up with the newer X-S20. 
> 
> It would be interesting to know if this could be improved at all with some tweaks! 
> 
> https://github.com/user-attachments/assets/08304648-457e-4688-8148-10265b592e05 

 _Originally posted by @cpkcpkcpk in [#181](https://github.com/gkoh/furble/issues/181#issuecomment-2827759647)_

gkoh, 2025-04-25:
That video is cool, I hadn't considered combining multi-connect with intervalometer!

May I confirm your intervalometer settings:
* Count: 999
* Delay: 700 ms
* Shutter: 30 ms

Further, it would help if you could obtain a serial console log:
* Connect M5StickC to PC
* Go to https://web-installer.furble.net
* Connect
   * Logs & Console

Then, repeat the multi-connect + intervalometer test.
You should get a timestamped trace log of `shutterPress()` and `shutterRelease()` for the cameras.
(Note, due to the recent power improvements, the serial log can get a bit garbled, which sucks, I can generate a build with it disabled if needed).

gkoh, 2025-04-25:
From prior poking about, I was expecting bluetooth latencies around 50ms, but it looks like 10x that, which doesn't seem right at all.

cpkcpkcpk, 2025-04-25:
Combining multi-connect with the intervalometer is awesome, as even most cheap wired intervalometers can only do 1 second minimum! I'm using this setup for 3D scanning, so not having any cable spaghetti and being able to press one button and shoot thousands of RAWs at 750ms interval on both cams is really great. Here is an example of my work - https://www.youtube.com/watch?v=zymBhfYVo8s

Those were the settings used in the video, but bumping up to 750ms seems a bit more reliable and didn't get any skipped frames using a count of 999.  I'm now using infinite count, but this did fail at some point and stopped shooting altogether.

I have attached the log below, but not sure if there is anything useful in there. Let me know if you want me to try again with another build.

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/19906350/esp-web-tools-logs.txt)

gkoh, 2025-04-25:
> Combining multi-connect with the intervalometer is awesome, as even most cheap wired intervalometers can only do 1 second minimum! I'm using this setup for 3D scanning, so not having any cable [...]

Wow, nice.
When I started this project I would never have guessed the possible applications.

> Those were the settings used in the video, but bumping up to 750ms seems a bit more reliable and didn't get any skipped frames using a count of 999. I'm now using infinite count, but this did fail at some point and stopped shooting altogether.

Hmm, infinite count fails?
Could you provide any details?
I'll investigate this as well, that shouldn't be the case.

> I have attached the log below, but not sure if there is anything useful in there. Let me know if you want me to try again with another build.

This is useful, I was after the timing trace.
On quick view, the numbers are reasonable, and the good news is there isn't some growth issue. The numbers remain fairly consistent over the 67 second run.

I also did a quick stats run.
For both cameras, the jitter between 'shutterPress' and 'shutterRelease' is pretty good, around 30-60ms.
There's a similar story for the delay.

This tends to suggest the controller and the bluetooth jitter aren't the primary cause.

I am now wondering if your initial thought was correct, the increased performance of the XS20 is a factor.

Hmm, you're shooting RAWs right?
How large are the files in MB?

cpkcpkcpk, 2025-04-25:
> Wow, nice.
> When I started this project I would never have guessed the possible applications.

It works great for my use case, and now I want more cameras to test with :D

> Hmm, infinite count fails?
> Could you provide any details?
> I'll investigate this as well, that shouldn't be the case.

It did just give up after a while the first few times I tested, but I probably had the shutter delay set too low, and maybe increased the shutter time too low or high too. Will see if I can reproduce, but infinite count seems ok with 750ms and 30ms.

> This tends to suggest the controller and the bluetooth jitter aren't the primary cause.

Interesting! Yeah, with 750ms I didn't notice any growth issue over 1000 shots. At 500ms the X-S10 started missing shots as it went on and and struggled to catch up

I'm shooting compressed lossless RAWs which are about 30MB. The X-S10 can shoot 3 RAWs per second in slow burst mode internally though at 333ms until the buffer is full, so I don't really understand why it would struggle with a slower interval over bluetooth.

gkoh, 2025-04-26:
> > Wow, nice.
> > When I started this project I would never have guessed the possible applications.
> 
> It works great for my use case, and now I want more cameras to test with :D

The limit for the ESP32 is 8.
If you do manage to get 8 cameras working I'd love to see it 🙂 

They don't all need to be the same either.
With the recent improved Canon EOS R and Nikon support, you should be able to mix and match.
 
> It did just give up after a while the first few times I tested, but I probably had the shutter delay set too low, and maybe increased the shutter time too low or high too. Will see if I can [...]

Thank you, please let me know if you do reproduce.

> Interesting! Yeah, with 750ms I didn't notice any growth issue over 1000 shots. At 500ms the X-S10 started missing shots as it went on and and struggled to catch up
>
> I'm shooting compressed lossless [...]

Hmmmmm.
I did a bit of a deep dive and I'm starting to think this is more to do with limits on the storage subsystem of the X-S10.
Whilst the burst mode can support 3 RAW FPS, the buffer memory is significantly faster than most SD cards and I don't think it could support a sustained 3 RAW FPS continuously.
To test this theory, it would interesting to just hold the shutter in CL and see if or when it starts to stall.

Officially, the X-S10 has support for UHS-I SD cards, supposedly meaning a maximum of 104MB/s.
Looking at random reviews, real SD card performance seems to range from 20MB/s up to 100MB/s.

At ~30MB per RAW, the SD card would need to support 60MB/s sustained write speeds at 500ms delay.

Out of curiosity, what is the make/model of the SD card being used in the X-S10?

Furthermore, @matthudsonau may be able to offer some insights here.

cpkcpkcpk, 2025-04-26:
Will let you know if I get my hands on more cameras!

Ahh yeah I think you are right about the SD cards.. I'm using a Sandisk Extreme UHS-I V30 card in the X-S10, and a Lexar 1800x UHS-II V60 in the X-S20. I just tried holding the shutter in CL 3fps on the X-S10 and it starts slowing down after maybe 30 shots, while the X-S20 can keep going until the card is full or battery runs out. Also tried shooting at around 2fps manually to the SD card and the X-S10 chokes even quicker.

I'll probably replace the X-S10 soon with another X-S20 or similar, but this works well enough for my needs at the moment and I'm happy that it works at all! Thanks a lot for looking into it.

gkoh, 2025-04-27:
> Will let you know if I get my hands on more cameras!

👍 
I have not being able to test more than 2 cameras.
It will be very interesting!

> Ahh yeah I think you are right about the SD cards.. I'm using a Sandisk Extreme UHS-I V30 card in the X-S10, and a Lexar 1800x UHS-II V60 in the X-S20. I just tried holding the shutter in CL 3fps on [...]

Ah!
Thank you for reporting back.
The V30 rating means it is rated for at least 30MB/s.
From your testing it sounds like it can only do 30MB/s given it can't even handle 2fps (60MB/s).

> I'll probably replace the X-S10 soon with another X-S20 or similar, but this works well enough for my needs at the moment and I'm happy that it works at all! Thanks a lot for looking into it.

You are most welcome.
This is the first feedback I've had for the multi-connect feature, so it's great to hear someone finding it useful.

I will close this issue now.

closed 2025-04-27 by gkoh