gkoh/furble #189: Support Canon EOS R GPS

gkoh, 2025-05-27:
Whilst Canon camera support is solid, we are missing GPS to complete the offering.
Add support to `furble` to send GPS data to Canon EOS R systems.

gkoh, 2025-06-03:
> Can’t access to RP, but R6m2 is available.

I think R6 should be fine.
If you wish to try, contact me at "guorong.koh AT gmail.com" and we can coordinate offline.

gkoh, 2025-06-20:
@Jerroder Unfortunately I did not hear from @hijae, thus your work is the most progress so far.
I do not have access to a Canon EOS R, but I am happy to assist.

> Edit: I got Bluetooth HCI logs from Android. The data is 32 bytes long, minus 1 byte for 0x04 at the beginning. However I don't see the content (latitude, longitude, date) in the data so it's probably compressed.

Good stuff!
32 bytes sounds about right.

With both Fujifilm and Sony, latitude and longitude are serialised as 4 bytes (`int32_t` ) representing the actual value multiplied by 10000000 (1e7).
The bytes could be little or big endian, you will need to crunch a few numbers to work out which.

There will be at least 7 bytes in date and time (2 for year, 1 each for month, day, hour, minute, second).

I started documenting the protocols, the Fujifilm location tag structure is here:
https://github.com/gkoh/furble/wiki/Protocol-Documentation#location-tagging

The Sony protocol is pending time availability.

Note that both Fujifilm and Sony location tag structures are fairly similar, so I'd guess (hope) Canon would not be too dissimilar.

gkoh, 2025-06-20:
> sorry, I'm busy right now and haven't gotten back to you, but I'll be available in middle of next week. If you reach out to the email in my profile, I'll check in with you when I have time.

Not a problem @hijae, I understand this is not a priority (even for me), do not feel pressured.
Let's see whether @Jerroder has more time (and hardware) than us to help 🙂

gkoh, 2025-06-20:
@Jerroder I just documented the Sony location tagging structure here:
https://github.com/gkoh/furble/wiki/Protocol-Documentation#location-tagging-1

gkoh, 2025-06-20:
Ah, I just remembered, I decoded Nikon's format, but have yet to actually use it, however, it is in the code:
https://github.com/gkoh/furble/blob/7f89d68af2371f5b656f1532ab13ffe5184bccaf/lib/furble/Nikon.h#L78

You will note that it is distinctly different, using "H'M'S" rather than degrees.

Jerroder, 2025-06-20:
Even the year (2025 is 07E9), I don't see it in the data sent by the android app, neither as 07E9 or E907, that's why I would think there's something more to it.

But I have time and an EOS R6 so we can have a look!

gkoh, 2025-06-20:
> Even the year (2025 is 07E9), I don't see it in the data sent by the android app, neither as 07E9 or E907, that's why I would think there's something more to it.
> 
> But I have time and an EOS R6 so we can have a look!

Hmm, interesting!
If you want any help, email me at 'guorong.koh AT gmail.com'.
I'm cognizant of breaching privacy by sharing location information on the public internet.

gkoh, 2025-06-20:
@Jerroder 
Oh, and thank you for the effort, very much appreciated 🙂

gkoh, 2025-06-21:
With the HCI log from @Jerroder, it looks like the GPS message format is:
```mermaid
packet-beta
  0: "0x04"
  1: "N|S"
  2-5: "Latitude"
  6: "E|W"
  7-10: "Longitude"
  11: "+|-"
  12-15: "Elevation"
  16-19: "Timestamp"
```
where:
  - "N|S" is ASCII representing north or south
  - "Latitude" is degrees as float32 in little endian
  - "E|W" is ASCII representing east or west
  - "Longitude" is degrees as float32 in little endian
  - "+|-" is ASCII representing negative or positive
  - "Elevation" is metres as float32 in little endian
  - "Timestamp" is seconds since epoch as uint32 in little endian

gkoh, 2025-06-21:
Further, (also thanks to @Jerroder), to enable GPS we need to:
- write `0x01` to Service UUID `00040000000010000000d8492fffa821` and UUID `00040002000010000000d8492fffa821`
- camera will respond with `0x02` on UUID `00040003000010000000d8492fffa821`

Then write the aforementioned GPS message to UUID `00040002000010000000d8492fffa821`

gkoh, 2025-06-28:
Based on the linked PR and testing by @Oaksen and @hijae we cannot support all functions at the same time with Canon EOS cameras.

Either the camera is paired in remote mode and you get 'shutter' and 'focus', or the camera is paired in smart device mode and you get 'shutter' and 'GPS'.
It seems we cannot support both 'focus' and 'GPS' at the same time.

closed 2025-11-04 by gkoh