gkoh/furble #192: Fujifilm - Unable to connect after successful pairing

mlstolk, 2025-06-07:
After performing a web install on a StickC-Plus2 (V3.2.1+1), I was able to bind to a Fujifilm X-T5 with (4.21). This worked straight with the first try and is repeatable after removing the binding, so far as expected.

However, connecting after having established the binding (camera is in the connect menu) it does not work. The X-T5 shows the smartphone symbol for apx. 4 seconds and then it disappears. The furble goes into retry in the Connecting screen (loops indefinitely). At each attempt, the camera shows the connection for that short period of time, the furble never is.

Any others with this problem? How can I debug what is preventing the furble from a successful connection to the camera?

gkoh, 2025-06-08:
@mlstolk Thank you for this interesting bug report.
I just tested with my X-T30 and did not experience this behaviour.

With the v3.2 release, I did upgrade the underlying bluetooth software.

If possible, please test with the previous version v3.1.2 via this permalink:
https://6902d65f.furble-web-installer.pages.dev/

If this works, it might be the updated bluetooth stuff.

> How can I debug what is preventing the furble from a successful connection to the camera?

The serial output can be accessed from the installer web page.
After connecting to the device, you get the options to either 'Install furble' or 'Logs & Console', choose the latter.

Due to the power saving stuff, the serial output is sometimes (regularly) corrupted due to the clocking changes, so you will likely see a lot of garbled output.

Unfortunately, I suspect there will be very little useful output without doing a full debug build.

gkoh, 2025-06-09:
> [@gkoh](https://github.com/gkoh) Thank you for helping out. The build that you suggested to install works straight away, so it seems the new bluetooth software version did break something here.

@mlstolk Thank you for testing!
When you get a chance, I would appreciate a test where I have reverted the bluetooth library.
The test release is here:
https://github.com/gkoh/furble/releases/tag/v3.2.2-rc0

Please use the 'Deployment URL', which is:
https://4bbc4509.furble-web-installer.pages.dev/

closed 2025-06-09 by gkoh

gkoh, 2025-06-09:
> [@gkoh](https://github.com/gkoh) Your test release (v.3.2.2-rc0) connects _without any issues_ as well. Thanks for guiding me through the process of isolating the issue.

Excellent, thank you very much for confirming, much appreciated.
I'm releasing v3.2.2 proper now.