gkoh/furble #193: Switchable delay after single trigger

mlstolk, 2025-06-08:
Great project! I think it would be very convenient if the single shot release allow for an easy (on/off switchable) delay of let's say 2 seconds . This delay could be configurable from the general settings menu as I presume it can be left static for most of the applications. After this delay has expired, the furble should hold the trigger button for long enough to acquire focus and trigger the actual release.
This way you can easily hide the trigger when making a tripod shot of yourself. With the current release function, the Furble is always in the frame (when your hands are). 

I'm aware this behavior can be emulated by setting the intervalometer on 2 photos with an interval of 2 seconds, but in that case:
- you would have one immediate photo every time as a 'bonus' (as the first of a series is taken without delay).
- you would have to reconfigure your interval settings back and forth every time you would like to 'misuse' it as a single delayed trigger.

Do more users see this as a convenient addition to single triggering?

_Also, accelerometer or microphone based triggering (with the same optional delay) would be great additions in the further future for certain event-based triggers (jump/smash/clap etc.)_

gkoh, 2025-06-09:
@mlstolk Thank you for the suggestions.

I can see the convenience of such of a delayed trigger feature, let me think about how to integrate it.

> *Also, accelerometer or microphone based triggering (with the same optional delay) would be great additions in the further future for certain event-based triggers (jump/smash/clap etc.)*

I have entertained a 'cheese' mode audio trigger since the very inception of this project. However I have not found a 'good' way to do it.
The ESP32 in the M5StickC series isn't well supported by Espressif's ESP-Skainet speech recognition framework and I have yet to find a reasonable alternative.

I've not considered motion based trigger. Could you elaborate on how this might be used?
(Perhaps double-tap?)

mlstolk, 2025-06-09:
@gkoh Great, that would be a great addition. Switching the camera's self-timer is an alternative for now, but it is limited and more hassle.

About the sensor-based triggering I can see various applications. They are all nice to haves, but I think it would open up the possibilities for many creative applications.

### • Trigger at impact
- Can be used to punch against the Furble when pocketed or while carrying gloves that make it impossible to press the button.
- Can be helpful with product photography when you want to take photos just when (or slightly after) an item hits something (e.g. the ground).

### • Trigger after acceleration-direction change
- Can be used when jumping (initial acceleration) and triggering at the top of the jump when the acceleration stops or reverses.

### • Sound-based trigger
- **Audio level**: Trigger when a certain audio level is reached.
  - This could be settable or teachable by recording an example and register the max detected volume (from clap, impact, mechanical impulse, etc.).
  - Advanced: Could be isolated to certain frequencies. And indeed an option for a double clap would reduce false positives.
- **Wake word**: Like the `'cheese'` attempt you tried earlier.

### • GPIO trigger
- Would be nice to allow an external sensor's signal to control the trigger.
- This would later let people expand to I2C, add sensors etc.
 
_With all of them, a configurable delay from the trigger would be essential to get the timing right._

Great to think along!

closed 2026-01-13 by gkoh

mlstolk, 2026-01-13:
Amazing, thank you!