gkoh/furble #195: Auto connect after power on

mlstolk, 2025-06-09:
@gkoh A minor enhancement suggestion for quicker operation.

Would it be possible to implement an auto-connect function that tries to connect and after connection goes to a quick shutter menu?
It could be enabled/disabled via the features menu.

So, after power on:
- it tries to connect to the (first) known camera. That menu should have a cancel button that leads you to the default menu (the menu that is currently shown after booting).
- after a successful connection, it immediately goes to the shutter menu.
- it would be convenient if that shutter menu has a power-off button for quick power-off as well, but it could be done by holding the power button.

This would make it very quick to use the Furble as a remote on the go.

gkoh, 2025-06-11:
This is possible.
I will have to investigate how to integrate into the existing boot sequence without hacking the code to pieces.

gkoh, 2026-04-25:
Apologies, I finally found time to implement this cleanly.
This is now release in [v3.8.2](https://github.com/gkoh/furble/releases/tag/v3.8.2).