gkoh/furble #201: ANy chance of getting the M5Tough made compatible

Grumpytech4, 2025-06-29:
THe M5Tough is a Core2 vintage controller with a touch screen, but no touch buttons. Furbi\le will load in the device, but the touchscreen is not operational so you are stuck with the greeting screen

The M5Tough would be an excellent device for those wanting a rugged device with a larger battery. THe GPS will also fit inside the device, giving a single unit with GPS and larger battery

gkoh, 2025-06-29:
@Grumpytech4 Good suggestion for additional hardware support.

The M5Tough appears to be supported by the underlying library used by `furble`, M5Unified.
So, yes, this should not be difficult to achieve (famous last words).

I do quite like the fully integrated and ruggedised nature of the unit.
However, that unit is outside by hobby budget (and also out of stock 😐).

If you have access to the M5Tough, GPS module and battery I would be willing to attempt blind support.
This would involve me guessing what I need to update and relying on you to test the builds.

If you would like to try, let me know.

Grumpytech4, 2025-06-29:
I have the earlier GPS UNIT and the M5Tough. I'm willing to give it a go and see if it can also be supported.
I do know the backlighting is not the same as the Core2 and have not been able to get it to dim even using M5's UIFLOW.

gkoh, 2025-06-29:
> I have the earlier GPS UNIT and the M5Tough. I'm willing to give it a go and see if it can also be supported. I do know the backlighting is not the same as the Core2 and have not been able to get it to dim even using M5's UIFLOW.

Excellent, thank you.
Let's give it a go 🙂

I do note that M5Stack's libraries aren't always great. And some hardware seems to just get ignored.

Grumpytech4, 2025-06-30:
The upload went without any problems, all functions work.

The touch screen works and all settings are editable, the touch zones often activate the zone below where you actually touch. If the finger is placed higher up on the screen the problem is not so apparent. I guess having nice touch squares like the shutter screen would help for those of us with clumsy big fingers

With no specific button on the M5Tough is there a way to lock the screen to stop erronous touch activations? The power button is unfortunately not readable from what I can tell.

Screen brightness works, which considering UIFLOW doesn't is quite a surprise. Inactivity timer also works

My main use with the larger battery will be to start it up and connect to the camera then place it in a backpack pocket for the day as I walk around, allowing the GPS to update whenever i use the camera, having a method to lock it in this state, even by disabling all touch permanently until reset with the power button would be a good compromise.

I'll use it over the next day or two and see how it goes

Thanks for the effort so far in this, it's much appreciated

Grumpytech4, 2025-06-30:
Forgot to ad that the Camera being used is the Fujifilm X-M5

gkoh, 2025-06-30:
> The upload went without any problems, all functions work.
>
> The touch screen works and all settings are editable, the touch zones often activate the zone below where you actually touch. If the finger [...]

Hmm, is this perhaps a touch screen calibration issue?

> With no specific button on the M5Tough is there a way to lock the screen to stop erronous touch activations? The power button is unfortunately not readable from what I can tell.

It's difficult to tell from the online images, but there is an actual power button, right?

If so, the schematic strongly suggests it is the same PMIC as the M5StickC Plus and wired the same.
There is a hack already in place to read that button, so I could port the hack over and try.

> Screen brightness works, which considering UIFLOW doesn't is quite a surprise. Inactivity timer also works
>
> My main use with the larger battery will be to start it up and connect to the camera then [...]

Great feedback. If you can confirm the physical existence of the power button, I can add the hack and spin a new test build.

> Thanks for the effort so far in this, it's much appreciated

No problems, happy to make this as useful as possible to all.

Grumpytech4, 2025-07-02:
Battery indicator does not move from half charge. With the battery flat to the point of turning off it still showed the same as having charged overnight. 

After an overnight charge the battery was at 3.82v which seems reasonable for a full charge and still showed half charge on the indicator. 

The Tough power management AXP192 has the ability to alter the charge rate for the battery. What is it currently set to charge at?

gkoh, 2025-07-02:
> There is a physical power button, looking at the diagrams it is connected directly to the PMU (AX192) there is no connection to the MCU.
> 
> When assembled the M5Tough obscures the pwr button with the housing, so I have fitted a flat panel mount waterproof one to the main body and wired it in parallel with the board one. It would be easy to wire the power switch to GPI36 or any other GPIO pin as well if required. M5 do not break out the PWR KEY to any pins which is a shame

Thank you for confirming.

Whilst you are correct, the PWR button state is (very) indirectly accessible via the AXP192 registers.
And M5 usefully abstracted the ability to read it (in a limited way) in the M5Unified library.

I have just cut [v3.3.0-rc11](https://github.com/gkoh/furble/releases/tag/v3.3.0-rc11) which leverages the same capability from the earlier M5StickC(Plus).

If this is certainly unworkable, then I have no problem with your GPIO wire up solution.

Grumpytech4, 2025-07-02:
PWR button hack does not work on the M5Tough or the Core2 to lockout the touchscreen

GPS appears to be operating the same on the Core2 and the Tough - with an update every 10s when looking at the data screen. Both devices connect and send GPS to FUJIFILM X-T4 and X-M5.

Battery charging is either very slow or not working correctly. I have a 1500mAh battery, it was charged for 14hours. I got approx 4 hours of runtime. I will investigate the actual charge current and see if I can capture the peak charge voltage as well

gkoh, 2025-07-02:
> PWR button hack does not work on the M5Tough or the Core2 to lockout the touchscreen

This is a Core2 and not a Core2 v1.1?
Unfortunately I only have Core2 v1.1 and it's another different PMIC 😐 

Given the spaghetti-ness of the M5Unified library it's not simple to determine whether Tough and/or Core2 AXP192 configuration is consistent with that of the M5StickC(Plus).

I'll keep poking when I get a chance.

> GPS appears to be operating the same on the Core2 and the Tough - with an update every 10s when looking at the data screen. Both devices connect and send GPS to FUJIFILM X-T4 and X-M5.

Oh, good, good!

> Battery charging is either very slow or not working correctly. I have a 1500mAh battery, it was charged for 14hours. I got approx 4 hours of runtime. I will investigate the actual charge current and see if I can capture the peak charge voltage as well

Hmmm, that certainly doesn't sound right.
Let me know if I can do anything.

If you're willing, getting a development environment going isn't too bad. Especially if you have some development experience.

Grumpytech4, 2025-07-02:
In Gold inside its written as **M5 Core II**  
The battery is now on this device and is charging a lot better, terminal voltage 3.85V already from flat, charge current is 364mA.  battery load is approx 90mA with GPS and fresh power up

on the Tough it charges at 90mA, battery load with a GPS and fresh power up is 100mA.

gkoh, 2025-07-02:
> In Gold inside its written as **M5 Core II**

OK, thank you.
It's quite messy with the way the M5Unified library handles these devices.
There's 2 Core2 devices, v1.0 and v1.1, each with different PMICs.
But there is only 1 type handler 😐 
And the library tries to be smart and work out which one, which makes it very difficult to know if it's doing the right thing.

> The battery is now on this device and is charging a lot better, terminal voltage 3.85V already from flat, charge current is 364mA. battery load is approx 90mA with GPS and fresh power up

Hmm, this is the 1500mAh in the Core2?
This certainly sounds like M5 did something inconsistent with either the PMIC wiring or soft configuration.

> on the Tough it charges at 90mA, battery load with a GPS and fresh power up is 100mA.

Either way, that load seems high.
I would expect ~50mA with GPS (~20mA for ESP32 + ~30mA for GPS).

gkoh, 2025-07-03:
I have just tagged v3.3.0 so we have a release with basic M5Tough working:
https://github.com/gkoh/furble/releases/tag/v3.3.0

> The current you expect, is that on the 5V USB supply or actual load on the nominal 3.7V battery?

Actual load on the battery.
I wrote up some of the power consumption investigation in the wiki:
https://github.com/gkoh/furble/wiki/Power-Consumption

Note that I relied upon the PMIC's current ADC for the measurements (as I really didn't want to disassemble the M5StickC).
The values I obtained seemed reasonably consistent with the datasheets so I didn't question them much.
If you are able to get measurements from an actual scope, that would be very welcome.

> The GPS is quoted around 30mA but it's a 5v device (150mW), using its own regulator for supply to the GPS chip
> 
> The battery load I would expect to be higher than this, with a 90% efficiency for the supply I would expect around a 45mA draw from the battery at 3.7v
> 
> I'll run some comparisons of current draw of the 2 devices (core and tough) with USB (with and without a battery) and battery alone as the source of power and put them in a table.
> 
> I'll also check what difference there is in current draw between it sitting idle at the wake-up screen vs connected vs sitting in auto reconnect looking for a camera.

It will be very interesting to see how your independent measurements with the values I obtained from the PMIC.

> With regard charging I'll put a scope on the battery and log the voltage over time and confirm exactly what the devices do.

👍

gkoh, 2025-07-03:
Apologies for spam commenting.
I noticed furble's M5 libraries are quite outdated, so I've updated and spun a test release:
https://github.com/gkoh/furble/releases/tag/v3.3.1-rc0

Nothing in the changes suggests it will fix the M5Tough power button press problem but one can hope.

Grumpytech4, 2025-07-11:
Some current draw figures for you using M5Tough USB powered vs the M5 core2 V1, no battery
GPS used for tests is the v1.1

current is average over 1min to get over the repeative peaks and spikes in power

first reading is the tough, second the core2 V1

Start up idle with GPS. 
92mA.    77mA
scan pressed, no camera on.   
156mA.  134mA
camera on, scan pressed. 
156mA. 134mA
connect to camera    
100mA. 83mA
disconnect  from camera.
92mA.    77mA

remove GPS and repeat

Power up with no GPS.              
46mA          30mA
scan pressed no camera on      
110mA        88mA 
camera on scan pressed.          
110mA.       88mA
connect to camera.                    
53mA.        27mA.     

turn on idle no gps.                     
46mA.       30mA
dim display to minimum.            
30mA.       30mA.       
max display brigtness -              
90mA.       64mA

**Battery charge check**

idle no battery                            
45mA.       30mA
battery connected                    
140mA.      404mA

**Auto reconnect enabled no GPS**

connected                                 
46mA.      27mA
turn off camera                       
110mA.      87mA
turn on camera                        
46mA.       27mA

**Auto reconnect with GPS that has a fix**
connected                            
100mA. 83mA
turn off camera                     
155mA.    134mA
turn on camera                    
100mA.       83mA

Some observations

Core2 draw lowers by 3mA when connected vs idle but Tough increased by 7mA
lowest setting on tough brightness is black - making restoration difficult if you change page
if you start scan before putting camera into the pairing mode and then start the camera it will not show in the scan unless you cancel and restart it

Biggest savings could be made by putting the GPS into idle for a period and not continuously polling for a camera on auto reconnect.

Grumpytech4, 2025-07-11:
[image]

M5tough with GPS turn on, connect to camera and then turn camera off for a period of time and then turn camera back on.

Grumpytech4, 2025-07-11:
[image]
M5core2 with GPS turn on, connect to camera then turn off camera for a period and turn back on

gkoh, 2025-07-16:
@Grumpytech4
Wow, thank you for expending the effort in obtaining actual traces.
These are roughly inline with the values I obtained from the PMIC itself, which is great.

> Some observations
> 
> Core2 draw lowers by 3mA when connected vs idle but Tough increased by 7mA

Wait, what?
Tough going into idle _increases_ current draw?
If so, it's almost certain that M5Unified and/or the underlying platform configuration is a bit broken.

In general, the Tough seems to suck quite a bit more power doing anything.

> lowest setting on tough brightness is black - making restoration difficult if you change page

The minimum brightness can be set specifically for M5Tough as needed.
Happy to experiment with this if desired.

> if you start scan before putting camera into the pairing mode and then start the camera it will not show in the scan unless
> you cancel and restart it

I see you've commented in #159, thank you.

> Biggest savings could be made by putting the GPS into idle for a period and not continuously polling for a camera on auto > reconnect.

I poked about and it would seem the GNSS Soc (ATGM336H-6N) may not have any idle mode.
Upon review of the schematic, it also strongly suggests VCC is hardwired to the 5V line, we can't even switch off the supply 😐 

So, that leaves #151, which should achieve _something_.
I need to think about how to refactor the control loop to do this cleanly.